luxai-qtrobot/QA #1: Preempting or canceling gestures/motor movements

RMichaelSwan, 2019-11-18:
Is there any support for this feature in the source code or interface? I believe it is possible to preempt commands with [some of the controllers often used with the ros_control package](http://wiki.ros.org/joint_trajectory_controller#Preemption_policy) so I was wondering if QT also supported this. It would be useful if some sensor indicated we should change our current movement plan.

apaikan, 2019-11-23:
Some robot actions such as playing gesture or audio files are not preemptable currently. 
It is in our release plan and will be available soon!

RMichaelSwan, 2019-12-03:
Are simple motor/joint movements preemptable right now? Our experimentation doesn't seem to indicate it, but perhaps we just aren't using the right method (currently just sending commands to services).