luxai-qtrobot/QA #2: Proper Full Shutdown disabled?

RMichaelSwan, 2019-11-18:
If the disable_interface option is set to true in /opt/ros/kinetic/share/qt_robot_interface/config/qtrobot-interface.yaml it seems the robot does not fully shutdown when the shutdown button is pressed or when the command is entered from the web interface. Specifically, I can still hear the Intel NUC's fan running even after the screen turns off. Is there a workaround for this or modification that could be made to the shutdown script?

apaikan, 2019-11-23:
It might happen sometimes motors do not go properly to their parking positions after shutdown. There should be some issues in `qt_motor` interface node which does not stop properly. 
However, your robot including RPI and NUC (if exists) DOES shutdown safely! 

**Notice :** Fan is always working until you unplug the robot from the power supply. So This is a normal behavior to hear fan after shutdown!

RMichaelSwan, 2019-12-03:
No issues on the motors, was just unsure if I should still be hearing the fan or not.