luxai-qtrobot/QA #3: QT WiFi redirects away from ROS wiki webpages

RMichaelSwan, 2019-11-18:
When the QT WiFi is connected to the internet and my own computer is then connected to the QT WiFi network, it redirects us to the config page (or some incorrect form of it) anytime we try to access pages on wiki.ros.org . Do you know what we can do to disable or modify that behavior?

apaikan, 2019-11-23:
Right! The reason is that RPI network is configured to route all traffics coming to http port 80 to 8080 using iptable.  

We will updated you briefly how to disable it!

apaikan, 2019-12-11:
Okay. here is the instruction how to disable port forwarding on QTrobot: 

- access QTrobot RPI via ssh. 
- navigate to `~/robot/autostarts` and open `start_qt_routes.sh` for editing (e.g. using vim/nano)
- comment this line `exec echo "$QT_USERNAME" | sudo -kS iptables -t nat -A PREROUTING -p tcp --dport 80 -j REDIRECT --to-port 8080`
- reboot the robot. 

**NOTICE**: Doing this, you need to access QTrobot RPI web-config interface using `http://qtrobot:8080` or `http://192.168.100.1:8080`

RMichaelSwan, 2020-01-08:
> Okay. here is the instruction how to disable port forwarding on QTrobot:
>
> * access QTrobot RPI via ssh.
> * navigate to `~/robot/autostarts` and open `start_qt_routes.sh` for editing (e.g. using [...]

I had to do this on the QTPC (NUC) instead of the RPi in order to get the desired effect (being able to browse internet through the QT### WiFi). I assume this is because the NUC is the one actually connected to the internet and the RPi is just passing on the web traffic to the NUC.

andrewpbstout, 2020-03-16:
Just a follow-up here to note that I had to do this (on QTPC, not RPi) in order to successfully `apt-get install` packages I needed for my applications.