luxai-qtrobot/QA #6: Wifi configuration page only shows available hotspots first time

RMichaelSwan, 2019-11-18:
For some reason QT's WiFi configuration webpage no longer shows any WiFi hotspots now that I have connected to a network. That said, I can tell the robot is definitely connected to the internet, but now I can't modify which network it is connected to from the web interface. Other than connecting to a network, I haven't really changed anything.

apaikan, 2019-11-23:
Let me clarify the QTrobot WIFIs and  web interface. As it has been explained in different places (#7) QTrobot  has an RPI and one NUC PC. 

- **RPI WIFI** is configured as Hotsopt with unique ssid for each robot (e.g. QT100). the web interface is accessible via `http://192.168.100.1:8080`
-  **NUC WIFI** is free and you can use it to connect to the internet either via it's Ubuntu desktop, terminal programs via ssh or using NUC web interface via  `http://192.168.100.2:8080` 

The reason that web interface of RPI does not allow you to change its WIFI configuration is intuitive: _you cannot access it remotely any more via its hotspot!_   QTrobot gave you three simple options to connect it to the internet: 

1. Using NUC WIFI
2. Using an external WIFI dongle attached to RPI USB port (back of the robot)
3. Using external Ethernet adapter attached to NUC USB-C port (see the attached picture)

**Notice:** It does not matter which option you use to connect the robot to the internet. The internet  will be always shared and routed between RPI and NUC!

**Notice:** If you connect external WIFI dongle to the RPI 's USB port, you need to reboot the robot and then it will be shown up in the RPI web interface!  

 
[image]

RMichaelSwan, 2019-12-03:
Thanks for the info!
So for some reason our NUC WiFi (192.168.100.2) does not allow us to configure any of its WiFi connections either (hotspot list is blank). Thoughts?

dkcodeguy, 2019-12-11:
Please check that you are using NUC web interface via http://192.168.100.2:8080, if you don't provide port 8080 it will redirect to RPI web interface. When you are using NUC web interface in the bottom left corner it should say QTPC. Please check also answer from @apaikan on #3 issue.