luxai-qtrobot/QA #9: read volume?

andrewpbstout, 2019-11-25:
Still in my initial poking around phase, but one of the first things I needed to do was change the robot's volume. (Out of the box it was a bit loud for my office...) I succeeded in using a rosservice call from bash to adjust the volume, so that's fine.

Is there a way to read what the volume is, i.e., getVolume as a counterpart to setVolume? Seems like that could be useful.

andrewpbstout, 2019-12-03:
Answered this myself before I posted it, I thought, but somehow it posted anyway. Sorry!

dkcodeguy, 2019-12-10:
For now we don't have a way to read the volume. We are planing to implement it in upcoming releases. You can always use 'alsamixer' from bash to adjust the volume.