luxai-qtrobot/QA #11: Microphone Topic

RMichaelSwan, 2019-12-03:
Are there plans for a microphone ROS topic or are we currently bound to the RPi in the head due to hardware/latency limitations?

andrewpbstout, 2019-12-04:
I'm also interested in the answer to this question. I'm evaluating wakeword providers, and some have greater compatibility for x86 than for RPi. I'm assuming doing wakeword detection on the NUC  would introduce unacceptable latency, but it would be nice to know if I'm wrong.

apaikan, 2019-12-11:
I thoroughly agree with you guys. Anyway there are few simple reasons why we did not implement custom Mic/audio streaming topic: 

1. First and most, **it already exists** in standard ROS under [audio_common](http://wiki.ros.org/audio_common) packages☺️ @dkcodeguy is providing a short tutorial on how to use it with QTrobot. 
2. Some of audio tools such as hotword detector need exclusive access to the microphone. So one should stop the ROS audio streaming node before running such a tool. 
3. QTrobot comes with [ReSpeaker Mic Array v2.0](http://wiki.seeedstudio.com/ReSpeaker_Mic_Array_v2.0/) which has 6 audio channels: 4 independent mic raw data, one processed audio for ASR and one merged playback channel. Although the ASR channel is most common used, depending on the user application, any of these configurations might be of interest.

dkcodeguy, 2019-12-12:
When using [audio_common](http://wiki.ros.org/audio_common/Tutorials/Streaming%20audio) you just need to configure capture.launch file to use proper device, so our mic is hw:1,0. Running audio_capture it will create "/audio/audio" topic and on that topic you can listen to the mic, read data etc. There is already tutorial on that page how to set it up. You can also try this [ros_respeaker](https://github.com/furushchev/respeaker_ros)