luxai-qtrobot/QA #12: silent recording from microphones

andrewpbstout, 2019-12-12:
Hi folks,

I'm trying to test Snowboy (http://docs.kitt.ai/snowboy/#running-on-pi) on QT, and I'm failing to get a test recording with 'rec test.wav'. I don't get an error, but when I play back the recording it's silent.

I've tried adding a ~/.asoundrc as they suggest, to no effect: 
`pcm.!default {
  type asym
   playback.pcm {
     type plug
     slave.pcm "hw:0,0"
   }
   capture.pcm {
     type plug
     slave.pcm "hw:1,0"
   }
}`

Any hints about how to configure to capture sound from the microphones?

dkcodeguy, 2019-12-13:
You can try with this `pcm.!default {  type asym  playback.pcm "plughw:0"   capture.pcm  "plughw:1" }
ctl.!default { type hw  card 1 }` and if your command doesn't work try this one `arecord -D plughw:1,0 --duration=5 test.wav`

andrewpbstout, 2019-12-13:
Thanks @dkcodeguy ! Your asoundrc doesn't seem to have had any effect, by your arecord command worked. 

It turns out that sox needs some environment variables to know the correct source and drivers: setting `export AUDIODEV=hw:1,0` and `export AUDIODRIVER=alsa` allowed me to get the same results with `rec` as I got with `arecord`.

But both are VERY quiet. I remember reading something suggesting that was common on RPis; I'll see if I can adjust the gain. alsamixer says that the capture device doesn't have any capture controls...what's The Right Way to adjust the gain? Or is that not actually my problem?

apaikan, 2019-12-17:
by VERY quite you mean completely silent (zero data) or very low gain data?
So far i know, for hot world detection or voice recognition, the audio gain is not important. in any case the audio algorithm normalizes the audio signal. 
Anyway, for audabilty purpose, you can increase 'AGCGAIN' Respeaker mic array. yoh may also need set 'AGCONOFF'. to do that you need to copy/download `tunning.py` to RPI to set the Mic parameters. please check the documentation here:http://wiki.seeedstudio.com/ReSpeaker_Mic_Array_v2.0/

andrewpbstout, 2019-12-17:
low gain--quiet, not completely silent. I was able to adjust the gain of the recording on the command line so I could hear for test purposes, and adjust the gain via PyAudio when testing Snowboy. Good to have the link to the Respeaker documentation, though!

Good news! After some hacking to get PyAudio to listen to the correct audio device, I got Snowboy's demo working on the robot. That's my plan for a hotword solution for now, but I'd love to know what other people are using.