luxai-qtrobot/QA #13: no network access from RPi

andrewpbstout, 2019-12-12:
As per https://github.com/luxai-qtrobot/QA/issues/5 my understanding was that once I connected QTPC to my wifi network, QTROBOT would have network access. That doesn't appear to be the case: I can ping (for example), ports.ubuntu.com from QTPC, but not from QTROBOT.

What else do I need to do to allow QTROBOT to access the network?

andrewpbstout, 2019-12-13:
Perhaps relatedly, from QTPC, 'ssh qtrobot@qtrobot' does not resolve--I have to do 'ssh qtrobot@192.168.100.1'.

andrewpbstout, 2019-12-13:
Ah! I have a hypothesis!

My network requires authentication from a webpage. I was able to do that from the QTPC, because I had a monitor plugged in. But the RPI is running headless (effectively), so I haven't done anything similar there. Could that be the issue, and do you have a suggestions for how to resolve it?

andrewpbstout, 2019-12-13:
And after 18 hours and a couple reboots, it seems to be working. hmmm.

RMichaelSwan, 2019-12-31:
I was having similar issues. The NUC would be connected to the internet but for some reason the RPi would not get the shared internet connection. Then randomly it would just start working. Still haven't figured out the cause.

rads560, 2020-02-12:
I'm also having this issue! Any suggestions for how to fix it?

dkcodeguy, 2020-02-13:
> Perhaps relatedly, from QTPC, 'ssh qtrobot@qtrobot' does not resolve--I have to do 'ssh qtrobot@192.168.100.1'.

If you want to ssh to RPI (QTROBOT) from QTPC you can do it as you said with 'qtrobot@192.168.100.1' or just 'ssh qtrp' without any password. Same thing goes for RPI if you want to ssh to NUC (QTPC) you can do it with IP 'qtrobot@192.168.100.2' or 'ssh qtpc'

> I was having similar issues. The NUC would be connected to the internet but for some reason the RPi would not get the shared internet connection. Then randomly it would just start working. Still haven't figured out the cause.

> I'm also having this issue! Any suggestions for how to fix it?

As per Internet on RPI. If you connected NUC (QTPC) to internet connection also RPI should have it. Please check instructions on how to connect your robot to internet connection on our [Documentation](https://docs.luxai.com/FAQ/Network/#2how-do-i-setup-wifi-on-qtrobot) page. After you connect always **reboot robot**. You can also try [removing port forwarding](https://docs.luxai.com/FAQ/Network/#1-how-to-disable-port-forward-on-qtrobot) from RPI.