luxai-qtrobot/QA #14: autostart explanations?

andrewpbstout, 2019-12-13:
This morning my robot started--unexpectedly, to me--saying "this is the happy face. I am also happy." I conclude that this is one of the built-in applications, reacting to a false detect of the happy face card (if I show it other emotion cards, it reacts accordingly). I unticked the "start_qt_emotion_app.sh" entry in the autostart menu and rebooted and that resolved the issue for the time being.

Is there a nice explanation somewhere of what the various entries in that menu are? the developer manual advises completely understanding the purpose of each script--where can I learn the purpose of each script so I don't accidentally turn off something critical?

I can't figure out what was triggering the false detect, but that does make me think twice about my plans, not yet enacted, to put a QT poster up on my wall...

apaikan, 2019-12-14:
okay, let me clarify some doubts:   

### What are autostart scripts?
 Autostart scripts are simply some bash scripts which are executed at robot startup time. These scripts prepare robot network, setup ROS environment, launch QTrobot motor and other controllers, setup log files and etc.  

### Where are they located? 
there are two sets of autostart scripts on QTrobot:
1. Autostart scripts on **RPI** 
2. Autostart scripts on **NUC**  

Both sets are located under `~/robot/autostart` folder. 

### How they are executed at startup time?
The scripts are run by linux [Cron](https://en.wikipedia.org/wiki/Cron) job scheduler. In fact, a specific corn job is configured to run `autostart_screens.sh` script at linux boot. The other scripts are launched by `autostart_screens.sh` and their output are redirected to the corresponding log files.  
[image]

To add your own script (e.g. `start_my_script.sh`) you can simply add the following line to the `autostart_screens.sh`: 

```
{
wait_for_network        
run_script "start_qtroutes.sh"
...
...
run_script "start_my_script.sh"
} &>> ${LOG_FILE}
```

### How can I access them?
You can access them via terminal of NUC/RPI or via QTrobot web interface for enable/disable them:

1. Web interface on RPI: `http://192.168.100.1:8080`
2. Web interface on NUC: `http://192.168.100.2:8080` 

### What are the main autostart scripts?
As I explained above, there are some _important_ scripts which prepare the robot network, ROS environment and launch the motor controller and other robot interfaces. All of these scripts are running on RPI and are **necessary** for functionality of QTrobot.  

#### Here are the main startup scripts on *RPI*:
- **`start_qt_routes.sh`**: prepares network routing between RPI and NUC for internet sharing and etc. 
- **`start_roscore.sh`**:  launches `roscore`
- **`start_qtpc.sh`**: power on NUC via wake-on-lan. 
- **`start_qt_motor.sh`**: launch `qt_motor` node including motor and gesture controllers.
- **`start_qt_robot_interface.sh`**: launch `qt_robot_interface` node which implements QTrobot emotion (face), audio, speech, behavior and other interfaces. 
-**`start_qt_webconfing.sh`**: launch QTrobot web config interface on RPI

#### Here are the main startup scripts on *NUC*:
The NUC pc is most convenient place to develop code and run your own autostart scripts and apps. 
Most of the default scripts are simply running the QTrobot open source demos and examples and can be safely disabled and/or modified. There is nothing especial running by autostart scripts on NUC except:
- **`start_qt_nuitrack_app`**: which run Nuitrack skeleton tracking and camera interfaces.  
- **`start_qt_webconfing.sh`**: launch QTrobot web config interface on NUC

### Where are the log files?
All programs/ROS nodes which are run by QTrobot autostart scripts redirect their standard output (info/war/error messages) to their corresponding log file. These log files can be found under `~/robot/autostart/logs` folder.

andrewpbstout, 2019-12-16:
Perfect. Thank you, @apaikan ! That explanation is ready to be put in the static developer documentation, if you haven't already done so. :-)