luxai-qtrobot/QA #16: Nuitrack face detection requires subject to be at least ~2 feet away

RMichaelSwan, 2020-01-08:
As mentioned in the heading, we need to be able to detect faces that are closer to QT (say ~1 foot away) in order to simulate face to face communication with a person (you don't usually stand 2+ feet away from people when talking to them). Deep learning based face detectors and even older techniques (Haar Cascade) don't have this limitation. Are there any plans to update /qt_nuitrack_app/faces to be more robust?

apaikan, 2020-01-21:
If your scenario, as you describe, is to detect faces within specific distance, why you dont simply filter out those which further 1 foot away? 

you have the skeleton tracking message including head/neck joint. you can use it to filter faces as you like. 

will this help?

apaikan, 2020-01-21:
I deleted my previous comments because i wrongly understood that you want to filter out some faces which are further than some specific distance! 

coming back to your question, you have access to the robot camera images. you can feed this image streams to any deep-learning or other type of face recognition library and use it with your application  For example you may take a look at Intel® [Movidius™ Neural Compute SDK](https://movidius.github.io/ncsdk/) which has plenty of examples including face recognition for different neural networks to be used with our without intel computational stick.

RMichaelSwan, 2020-01-21:
I am already looking into some deep learning model solutions, I was just hoping that some code or library might be readily available or that we could get some support from the Nuitrack team. One thing to note is that most deep learning models just use 2D RGB data and do not make use of depth. There is also the issue of inference time--Nuitrack runs very fast (~80% cpu usage on one core while maintaining 30 FPS at a good resolution) while some of the most efficient face detectors I have found struggle to get more than 5-10 FPS (with max CPU usage on the NUC) at low resolutions on our CPU bound setup.

The Movidius resource ([especially the app zoo](https://github.com/movidius/ncappzoo)) is one I wasn't aware of though, I'll try to see how it performs

andrewpbstout, 2021-03-04:
Hi. I'm re-animating this thread just to ask @RMichaelSwan what you settled on, and ask the wider community (if anyone else is listening) what folks are using for face reco on QT. 

I just started playing with face detection about half an hour ago. So far I've just done a `rostopic echo /qt_nuitrack_app/faces` and moved around in front of the robot. I can see that it's working, but detection seems a little spotty. Maybe because I'm in a small room and can only get about 2 feet away from the robot, and the usual distance is more like 18 inches? 

I do appreciate that when it does see me it thinks I'm younger than I actually am, but I also seem to have resting emotion_angry face. ;-)

Just wondering if most folks are using the stock face detection from qt_nuitrack, or if there's another package that works well out of the box.

RMichaelSwan, 2021-03-04:
@andrewpbstout there are a bunch of neural networks and algorithms for face recognition/detection and emotion detection, though it's definitely a challenge to find ones that run efficiently when CPU bound (as many use GPU). My group decided to roll our own solution using open source libraries. The one we settled on for face detection (only) was dlib, which runs quite efficiently and doesn't have the distance problem. We also tried FaceNet, though I don't recall the results there (I believe it's a little slower, but more robust).

In any case, our code is open source, it runs on QTrobot. We opted for using Python3 which is a little harder to use on ROS Kinetic, but it works. We also put some Docker containers together which also do the trick. [Here's our dlib detector](https://github.com/interaction-lab/HARMONI/blob/develop/harmoni_detectors/harmoni_face_detect/src/harmoni_face_detect/detector.py) and our codebase (feel free to make use of it/contribute!): https://github.com/interaction-lab/HARMONI

andrewpbstout, 2021-03-05:
@RMichaelSwan Thanks for the reply! HARMONI sounds like it could be really useful--I'm eager to check it out in more detail, although I might have limited bandwidth to do so in the short term. (I wish it had existed/I had known about it a year and a half ago!) I'd love to read your paper.

RMichaelSwan, 2021-03-06:
Heh, we started development on it about a year and a half ago, so that might have been difficult ;). I'll double check with the authors, but it should be okay to share the paper at this point.

In the near term, if you want to just have something going quickly, you could probably just tear out the detector package we have and shift some code around to work with whatever image source you have, though I would have to make a biased recommendation to adopt HARMONI entirely instead :)