luxai-qtrobot/QA #24: QT Pi Hotspot not working

micolspitale93, 2020-02-24:
During a recording session, the nuc crashes. For restarting QT, we just unplug and plug in again. 

When we restarted the robot, it seems that the pi hotspot was not working anymore. Actually until the nuc is not switch on, the hotspot is working, but as soon as the nuc turns on it just stopped working. 

How could we solve this issue? 
Thank you

dkcodeguy, 2020-02-25:
Hi,
Can you explain what kind of "recording session"?
For restarting "power cycling" QT you should first turn off QTrobot by pressing the power button, power off with ssh or from web control panel and after that unplug it. As it was explained in the [documentation](https://docs.luxai.com/user-manual/#power-off) it's not recommended to power off QTrobot by cutting the power supply off.

dkcodeguy, 2020-03-05:
Hi @micolspitale93 can you update me if everything is ok?
Thanks

micolspitale93, 2020-03-05:
I’m so sorry for the delayed reply. During the “recording session”, I was just recording a video, and the memory saturated. This causes the crash of the nuc. I was not able anymore to switch off or restart the nuc neither with the power button (I could not even ssh). So I decided to unplug the robot and reboot it.As soon as the robot rebooted, the QT network was just working when the nuc was off. Right now, everytime the nuc started the QT hotspot stops working. I was not able to solve the problem.  I hope this explanation is more clarifying.    From: dkcodeguySent: 05 March 2020 02:46To: luxai-qtrobot/QACc: Micol Spitale; MentionSubject: Re: [luxai-qtrobot/QA] QT Pi Hotspot not working (#24) Hi @micolspitale93 can you update me if everything is ok?Thanks