luxai-qtrobot/QA #25: how to change the speech language of QTrobot in my own PC

ZouJennie, 2020-03-05:
I did already ROS_IP and ROS_MASTER_URI to connect with the qtrobot and i can see all topic list. 
I tried to change the speech language of qtrobot with the commande : rosservice call /qt_robot/speech/config "fr-FR 0 0" in the terminal of my PC
then i got a error such as : ERROR: Unable to load type [qt_robot_interface/speech_config].
Have you typed 'make' in [qt_robot_interface]?

So what can i do?

dkcodeguy, 2020-03-05:
Hi @ZouJennie welcome to the QTrobot community :) 
Can you please run the same command on QTPC and let me know what results do you get?

ZouJennie, 2020-03-05:
> Hi @ZouJennie welcome to the QTrobot community :)
> Can you please run the same command on QTPC and let me know what results do you get?

rosservice call /qt_robot/speech/config "fr-FR 0 0"
ERROR: Incompatible arguments to call service:
Not enough arguments:
 * Given: ['fr-FR 0 0']
 * Expected: ['language', 'pitch', 'speed']
Provided arguments are:
 * fr-FR 0 0 (type str)

dkcodeguy, 2020-03-05:
I guess your not calling the service from command line properly. 

your command should look like this: 
```bash
$ rosservice call /qt_robot/speech/config "language: 'fr-FR'
pitch: 0
speed: 0" 
```
**Notice** that this is standard way of calling ROS services from command line.  to make sure you are typing it properly press `TAB` few times after `rosservice call /qt_robot/speech/config` to get autocomplete of the available parameters.

ZouJennie, 2020-03-05:
> I guess your not calling the service from command line properly.
> 
> your command should look like this:
> 
> ```shell
> $ rosservice call /qt_robot/speech/config "language: 'fr-FR'
> pitch: 0
> speed: 0" 
> ```
> 
> **Notice** that this is standard way of calling ROS services from command line. to make sure you are typing it properly press `TAB` few times after `rosservice call /qt_robot/speech/config` to get autocomplete of the available parameters.

With your command it works on QTPC. But i want to know is it possible to call this ROS service on my PC? And need i change the language every time after turn on the robot?

qtrobot, 2020-03-09:
@ZouJennie It is possible to run the commands/services from your PC. The missing part is some QTrobot ROS message/service header files. We have included them already on QTPC. 

@dkcodeguy  will send you the files with instruction on how to install them on your ROS.

dkcodeguy, 2020-03-09:
@ZouJennie To Install QTrobot header files please follow the steps on our documentation page under [Installing ROS](https://docs.luxai.com/user-manual/#installing-ros) ***6. Install QTrobot ROS header files***

**Notice** this you need to do on your PC.

ysl208, 2023-05-14:
Hi, I tried this but the service only returns `True` if I set language to `en-US` but not for any other (I tried `en-GB`, `de-DE`, `fr-FR`, `es-ES`). When I check in `~/robot/data/tts-lang/` there are a lot more languages available. Any help?