luxai-qtrobot/QA #29: Gesture Kinematic Diagram

VR-Nic, 2020-03-23:
Hey folks,

Is there any documentation for the gesture system with the axis and limits for each QTrobot part? I'm looking for something like a kinematic diagram that shows the limits, axis, and directionality for each moveable joint on the QTRobot.

Thanks,
Nic

qtrobot, 2020-03-25:
Hi @VR-Nic ,
Here are some information you may find it useful for your question: 

## Gesture files 
the gesture are simply xml files containing robots joint waypoints and their timestamps. 
They are located under `~/robot/data/gestures/` folder in QTRP (head) board.  here is an example: 
```xml
<?xml version="1.0" encoding="utf-8" standalone="yes" ?>
<gesture>
    <name>QT/bye</name>
    <parts>
        <part>right_arm</part>
        <part>left_arm</part>
    </parts>
    <duration>5.54</duration>
    <waypoints count="97">
        <point time="1558971402671926152">
            <LeftElbowRoll>-33.50</LeftElbowRoll>
            <LeftShoulderPitch>89.90</LeftShoulderPitch>
            <LeftShoulderRoll>-62.90</LeftShoulderRoll>
            <RightElbowRoll>-31.90</RightElbowRoll>
            <RightShoulderPitch>-88.60</RightShoulderPitch>
            <RightShoulderRoll>-59.30</RightShoulderRoll>
        </point>
        <point time="1558971402704626777">
            <LeftElbowRoll>-33.50</LeftElbowRoll>
            <LeftShoulderPitch>89.90</LeftShoulderPitch>
            <LeftShoulderRoll>-64.80</LeftShoulderRoll>
            <RightElbowRoll>-30.60</RightElbowRoll>
            <RightShoulderPitch>-88.60</RightShoulderPitch>
            <RightShoulderRoll>-60.60</RightShoulderRoll>
        </point>
...
```
## QTrobot Joints configuration 
in QTRP, under `/opt/ros/kinetic/share/qt_motor/` there are motor controller config files: 

- **`qtrobot-hardware.yaml`**: contains each motor configuration such as joints limit, PID, calibration offsets, etc. 
  
- **`qtrobot-controller.yaml`**:  contains QTrobot controller interface configuration such as `joint_state`, `gesture`, etc.  These interface are implemented using  ROS control system. 

## QTRobot URDF
  You can find the QTrobot URDF fils in the following links: 
 - [qtrobot.pdf](https://github.com/luxai-qtrobot/QA/files/4381012/qtrobot.pdf)
 - [qtrobot_urdf.zip](https://github.com/luxai-qtrobot/QA/files/4381016/qtrobot_urdf.zip)

You can also load them in Gazebo to visualize joint configurations and test them: 
```
$ roslaunch urdf_tutorial display.launch model:=qtrobot.urdf
```
you need to copy the `qtrobot.urdf` file to the proper folder in your PC (where you have installed Gazebo). Check the `display.launch` to find the path. 

Hope this can help you :)

qtrobot, 2020-04-22:
Please also see our latest tutorial on how to use ROS MoveIt with QTrobot:  https://github.com/luxai-qtrobot/tutorials/tree/master/examples/motors_moveit