luxai-qtrobot/QA #30: Nuitrack UVCVideoCapture::open ERROR: can't find device

abouttier, 2020-04-07:
Hello,
for a test we connected another webcam and tried droidcam ...
since I have an error which is present when I want to launch nuitrack.

[image]

I managed to go back on the drivers and softs used but I still have this error.
How can I do ?
Re-install the entire nuitrack?

uninstall and reinstall the realsense camera?
I have tried many things but little success!
thank you in advance

abouttier, 2020-04-08:
Hello,
after verification, all the applications work, demos work.
After doing some research, I found a post that explains why nuitrack_app displays these messages.
Is it okay for you ?

https://community.nuitrack.com/t/solved-cant-find-device-error/1422

qtrobot, 2020-06-03:
Hi @abouttier 
This is  really strange. Because usually QTrobot comes with Nuitrack already installed and properly configured!  
Would you let us know what exactly happened?