luxai-qtrobot/QA #31: can we add head position to move_group of moveit

ZouJennie, 2020-05-13:
Hello I'm the intern of projet iReCheck co-writter. I'm working on the moveit to control the QT to do some behaviours. But I realize there are only two move group in the package "motor moveit" without the group of head yaw and head pitch. I'd tried to control the head with ros publisher but I can no more find the rostopic "/qt_robot/head_position/command" which is indicated in the tutorial.  
So how can I add the head to the move_group? Or do you know why i can't find this rostopic? I'm sure that two PC of robot and my PC are all connect to the same master.

ZouJennie, 2020-05-18:
> Hello I'm the intern of projet iReCheck co-writter. I'm working on the moveit to control the QT to do some behaviours. But I realize there are only two move group in the package "motor moveit" [...]

I saw in the file qtrobot.srdf there is a groupe named "head" but when i lanch the moveit_qtrobot.launch there are only two planning group: left and right arm. As while when i try the function "get_group_name", there are only two group too. How can i find the head group?

qtrobot, 2020-06-23:
hi @ZouJennie Sorry for the late reply. The challenge is that ROS move_it package has some issues with iKFast solver for 2DoF (LookAt) interface. We are working on that and let you know.