luxai-qtrobot/QA #32: loop an emotion?

andrewpbstout, 2020-06-01:
Hi--

(I can't find this documented at https://docs.luxai.com/api/)

Is there a way to tell the "emotion" (face display) system to loop an "emotion" animation until otherwise instructed? I assume so, since that's what it does for neutral_state_blinking; I'd like to loop the `talking` animation for the duration of a service call to '/qt_robot/speech/say'.

Thanks in advance!

qtrobot, 2020-06-03:
Hi @andrewpbstout 
I believe this is already implemented in QTrobot Behavior interface (https://docs.luxai.com/api/#45-behavior-interface). 

You need to use `/qt_robot/behavior/talkText` service or subscriber.

andrewpbstout, 2020-06-03:
Ah, perfect, I was just looking in the wrong place. Thanks!

closed 2020-06-25 by qtrobot