luxai-qtrobot/QA #33: respeaker_ros startup issue

andrewpbstout, 2020-06-01:
Anyone here using https://github.com/furushchev/respeaker_ros ? I'm using it, mostly with success, but most of the time the first time I launch it, it fails to correctly find the respeaker device. If I cancel and restart, it goes fine.

Anyone have a solution for this problem? (I'm not finding a troubleshooting doc for respeaker_ros...)

dkcodeguy, 2020-06-03:
Hi @andrewpbstout,
We don't use this, but we can test it and let you know if we find a solution.

andrewpbstout, 2021-09-02:
Just wanted to bump this--I'm still having this issue, and I need to fix it. I've tried updating the firmware on the respeaker and regenerating the config file, to no effect.

andrewpbstout, 2021-09-02:
Here's my output the first time I launch:

```
qtrobot@QT140:~$ roslaunch respeaker_ros respeaker.launch
... logging to /home/qtrobot/.ros/log/3f68f30a-0aa8-11ec-845b-b827ebb335e7/roslaunch-QT140-3272.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.100.1:38599/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /speech_to_text/language: en-US
 * /speech_to_text/self_cancellation: True
 * /speech_to_text/tts_tolerance: 0.5

NODES
  /
    respeaker_node (respeaker_ros/respeaker_node.py)
    speech_to_text (respeaker_ros/speech_to_text.py)
    static_transformer (tf/static_transform_publisher)

ROS_MASTER_URI=http://192.168.100.1:11311

process[respeaker_node-1]: started with pid [3289]
process[static_transformer-2]: started with pid [3291]
process[speech_to_text-3]: started with pid [3292]
[INFO] [1630611147.801433]: Initializing Respeaker device
[ERROR] [1630611149.583799]: action 'sound_play' is not initialized.
[INFO] [1630611158.066225]: Respeaker device initialized (Version: 16)
[WARN] [1630611158.861267]: Failed to find respeaker device by name. Using default input
[WARN] [1630611158.863277]: 32 channel is found for respeaker
[WARN] [1630611158.864705]: You may have to update firmware.
[INFO] [1630611158.866141]: Using channels [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31]
```

andrewpbstout, 2021-09-02:
If I hit ^C to kill it and the run again, it works:
```
qtrobot@QT140:~$ roslaunch respeaker_ros respeaker.launch
... logging to /home/qtrobot/.ros/log/3f68f30a-0aa8-11ec-845b-b827ebb335e7/roslaunch-QT140-3515.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.100.1:39587/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /speech_to_text/language: en-US
 * /speech_to_text/self_cancellation: True
 * /speech_to_text/tts_tolerance: 0.5

NODES
  /
    respeaker_node (respeaker_ros/respeaker_node.py)
    speech_to_text (respeaker_ros/speech_to_text.py)
    static_transformer (tf/static_transform_publisher)

ROS_MASTER_URI=http://192.168.100.1:11311

process[respeaker_node-1]: started with pid [3532]
process[static_transformer-2]: started with pid [3534]
process[speech_to_text-3]: started with pid [3545]
[INFO] [1630611531.437069]: Initializing Respeaker device
[ERROR] [1630611534.233465]: action 'sound_play' is not initialized.
[INFO] [1630611541.703970]: Respeaker device initialized (Version: 16)
[INFO] [1630611541.801501]: Found 2: ReSpeaker 4 Mic Array (UAC1.0): USB Audio (hw:1,0) (channels: 6)
[INFO] [1630611541.804107]: Using channels [0, 1, 2, 3, 4, 5]
```

andrewpbstout, 2021-09-02:
(Sorry for spamming the forum, but I'm just going to bring you all with me as I try to get to the bottom of this. Any help appreciated!)
It turns out I can get similar results using this get_index.py script:
```
import pyaudio
 
p = pyaudio.PyAudio()
info = p.get_host_api_info_by_index(0)
numdevices = info.get('deviceCount')
 
for i in range(0, numdevices):
        if (p.get_device_info_by_host_api_device_index(0, i).get('maxInputChannels')) > 0:
            print "Input Device id ", i, " - ", p.get_device_info_by_host_api_device_index(0, i).get('name')
```

I reboot the robot, and then run this script twice in succession:
Note that the first time I run it, it doesn't detect the ReSpeaker:
```
Input Device id  2  -  pulse
Input Device id  3  -  default
```
but the second time it does:
```
Input Device id  2  -  ReSpeaker 4 Mic Array (UAC1.0): USB Audio (hw:1,0)
Input Device id  4  -  pulse
Input Device id  6  -  default
```

Full output (I think the ALSA error spew is a distraction):
```
qtrobot@QT140:~/respeaker_utils$ python get_index.py
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.front.0:CARD=0'
ALSA lib conf.c:4292:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4771:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4292:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4771:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4292:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4771:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4292:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4771:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4292:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4771:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4292:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4771:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4292:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4771:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4292:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4771:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4292:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4771:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
[... 53 lines cut ...]
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4292:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4771:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
[... cut ...]
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
connect(2) call to /tmp/jack-1000/default/jack_0 failed (err=No such file or directory)
attempt to connect to server failed
Input Device id  2  -  ReSpeaker 4 Mic Array (UAC1.0): USB Audio (hw:1,0)
Input Device id  4  -  pulse
Input Device id  6  -  default
```

andrewpbstout, 2021-09-02:
...and if I've already run get_index.py once, starting respeaker_ros works the first time.

dkcodeguy, 2021-09-03:
Hi Andrew,

We have our version of respeaker_ros which you can find [here](https://github.com/luxai-qtrobot/software/tree/master/apps/qt_respeaker_app). We are using it on our latest version of QTrobot which has ubuntu 20 on QTRP and is running with python3. We also added tuning parameters for noise reduction and gain, you can then customize how ever you like it. You will also not see anymore this alsa printout. You can try and check the code, I think functionality is similar as the one you get from respeaker_ros.

On QTRP you will need just [audio_common_msgs](http://wiki.ros.org/audio_common_msgs?distro=noetic), but if you are using respeaker_ros i think you should already have it.
If you don't have python3 you can try with virtual environment and install all requirements in that. The only thing you would need to change is autostart script. If you need help with that we can arrange a call for you to set it up.

This alsa printout is nothing to worry about. We found that running with sudo helped us with similar issue. You can try that with respeaker_ros.

andrewpbstout, 2021-09-03:
Thanks @dkcodeguy , I'll check this out. My QT is still running ubuntu 16 and python 2--will this work without upgrading everything?

dkcodeguy, 2021-09-06:
You can install python3 and use python3 virtual environment for the project. If you need help with this let us know.

FrederikeDurow, 2024-04-08:
Hey @andrewpbstout, it has been a while since your issue, did you solve it? 
I get the same on Ubuntu 22.04, with ROS2 humble. 
When I call the get_index.py file I always see the respeaker, but it detects the sound very wierdly because of the 32 channels. On a different computer with the same code it works.