luxai-qtrobot/QA #36: Robot IP connection

dsafaei69, 2020-06-22:
I have a problem with QT, after I connect to the QT wifi and set everything and export the URI variable to my profile. I can get the list of topics (rostopic list) from my machine.
However, when I run a topic query on the client-side, I observe no actions from the robot. I checked inside master.log on the robot side, my query was logged but the robot did not show any actions.

Here is the log for speech, for example:
 
 
[rosmaster.master][INFO] 2020-06-19 14:32:24,486: +SERVICE [/rostopic_18727_1592761450442/get_loggers] /rostopic_18727_1592761450442 http://213.55.220.54:44637/
[rosmaster.master][INFO] 2020-06-19 14:32:24,514: +SERVICE [/rostopic_18727_1592761450442/set_logger_level] /rostopic_18727_1592761450442 http://213.55.220.54:44637/
[rosmaster.master][INFO] 2020-06-19 14:32:24,520: +PUB [/qt_robot/speech/say] /rostopic_18727_1592761450442 http://213.55.220.54:44637/
[rosmaster.master][INFO] 2020-06-19 14:32:24,559: publisherUpdate[/qt_robot/speech/say] -> http://192.168.100.1:34979/ ['http://213.55.220.54:44637/']
[rosmaster.master][INFO] 2020-06-19 14:32:24,561: publisherUpdate[/qt_robot/speech/say] -> http://192.168.100.1:34979/ ['http://213.55.220.54:44637/']: sec=0.00, result=[1, '', 0]
[rosmaster.master][INFO] 2020-06-19 14:32:38,315: -PUB [/qt_robot/speech/say] /rostopic_18727_1592761450442 http://213.55.220.54:44637/
[rosmaster.master][INFO] 2020-06-19 14:32:38,316: -SERVICE [/rostopic_18727_1592761450442/get_loggers] /rostopic_18727_1592761450442 rosrpc://213.55.220.54:36665
[rosmaster.master][INFO] 2020-06-19 14:32:38,316: -SERVICE [/rostopic_18727_1592761450442/set_logger_level] /rostopic_18727_1592761450442 rosrpc://213.55.220.54:36665
[rosmaster.master][INFO] 2020-06-19 14:32:38,384: publisherUpdate[/qt_robot/speech/say] -> http://192.168.100.1:34979/ []
[rosmaster.master][INFO] 2020-06-19 14:32:38,386: publisherUpdate[/qt_robot/speech/say] -> http://192.168.100.1:34979/ []: sec=0.00, result=[1, '', 0]

Could you help me how can I fix the problem?

qtrobot, 2020-06-23:
Hi @dsafaei69. Can you confirm that you have set up `ROS_IP` variable correctly?   
The instruction is given on our tutorial guide: [Configure ROS environment for QTrobot](https://docs.luxai.com/user-manual/#7-working-directly-on-qtpc)

The `ROS_IP` should be set in your bash along with other indicated environmental variables and should refer to your PC IP address after connecting to the QTrobot  WIFI.  please check it as bellow: 
```
$ echo $ROS_IP
10.42.0.x
```
If yes, what do you get by running the following command: 
```
rostopic pub /qt_robot/speech/say std_msgs/String "data: 'Hello!'"
```

you can also let us know your ROS version and configuration as follows: 
```
$ rosversion -d
$ env | grep ROS
```

dsafaei69, 2020-06-25:
I Checked the steps again and everything was correct but when I run an example of robot actions it does nothing.
my ROS version is kinetic 

$ env | grep ROS
ROS_ROOT=/opt/ros/kinetic/share/ros
ROS_PACKAGE_PATH=/home/chili/catkin_ws/src:/opt/ros/kinetic/share
ROS_MASTER_URI=http://localhost:11311
ROS_PYTHON_VERSION=2
ROS_VERSION=1
ROSLISP_PACKAGE_DIRECTORIES=/home/chili/catkin_ws/devel/share/common-lisp
ROS_DISTRO=kinetic
ROS_ETC_DIR=/opt/ros/kinetic/etc/ros

qtrobot, 2020-06-25:
@dsafaei69 can check first that everything is working properly on the robot by following these steps: 

-  ssh to QTrobot (`ssh qtrobot@192.168.100.1`) and test speech or emotion interface (e.g. `rostopic pub /qt_robot/speech/say std_msgs/String "data: 'Hello'"`) from ssh terminal.
-  ssh to QTPC (`ssh qtrobot@192.168.100.2`) and test speech or emotion interface (e.g. `rostopic pub /qt_robot/speech/say std_msgs/String "data: 'Hello'"`) from ssh terminal. 
- Double check if your laptop/pc ip address is actually `10.42.0.153`. 

Are you sure that you are calling the QTrobt speech interface correctly via ROS topic?  use `<TAB>` to autocomplete the command line to avoid misspelling!  Avoid also `!` character in speech interface from command line!

dsafaei69, 2020-06-25:
about the commands yes, I am sure; I can get the rostopics on the PC side and I wrote the command by using tab, furthermore, when I connected to the robot by <ssh qtrobot@qtrobot> and ran the same command on the robot directly it worked and robot reacted correctly. As I mentioned in my first msg when I checked the logs on the robot it received the commands the same as when I run them on the robot directly but it doesn't have any reaction to the commands.

qtrobot, 2020-06-26:
That's good. It seems that the robot is working fine and the communication between QTrobot and QTPC is fine too. 
Looking into your log file, your laptop IP is `213.55.220.54`. It does not seem to be in the same network of `10.42.0.153`. 

- How did you connect to the robot? Did you connect your laptop to QTrobot wifi (ssid: `QTxxx`)? 
- can you check if you can read some topics from robot (e.g. `/qt_robot/joints/state`)
- What about calling the speech interface using ROS service (`rosservice call /qt_robot/speech/say "message: 'Hello'"`)?

dsafaei69, 2020-07-13:
I checked them all.
Yes, I connected to QTxxx wifi. I have access to the topics and services on my PC but I cannot run a python file on QT.
Is it possible to have an online meeting to solve the problem?
I already checked the probable issues but I cannot find what would be the problem.

closed 2020-07-20 by dsafaei69