luxai-qtrobot/QA #43: Could we build emotion ourselves?

ZouJennie, 2021-04-20:
Hello,

During the advancement of our research project, we found that the existing expressions of emotion were not enough to meet our needs. We want to know if there is a tool that can create the same style of expressions. 

Thank you

dkcodeguy, 2021-04-21:
Hello Zoe,
I am afraid we have no tool enabling you to change/extend the existing faces of QT. But if you let us know what other facial expressions you would need, we may be able to develop and include them in our next releases.

You may however want to develop your own faces for QTrobot. Here you have full control over the QTrobot’s LCD and you can change the QTrobot’s default faces with your own custom images or videos. If this is the approach you are thinking of, please let me know to provide you with detailed instruction on how to run your own images/videos on QT’s face.

ZouJennie, 2021-05-04:
> Hello @ZouJennie,
> I am afraid we have no tool enabling you to change/extend the existing faces of QT. But if you let us know what other facial expressions you would need, we may be able to develop and include them in our next releases.
> 
> You may however want to develop your own faces for QTrobot. Here you have full control over the QTrobot’s LCD and you can change the QTrobot’s default faces with your own custom images or videos. If this is the approach you are thinking of, please let me know to provide you with detailed instruction on how to run your own images/videos on QT’s face.

Thank you for your answer. We have no specific requirements yet. After we confirm, we will contact you to inquire about related issues.

aerith-zkelov, 2021-09-12:
> Hello @ZouJennie,
> I am afraid we have no tool enabling you to change/extend the existing faces of QT. But if you let us know what other facial expressions you would need, we may be able to develop [...]

Hi i'm interested to try this out? May i know how to display images/videos on qt's face?

dkcodeguy, 2021-09-13:
Hi @aerith-zkelov,
The simple way is to use qtrobot emotion interface, which can only play .avi files. Copy your video to `/home/qtrobot/robot/data/emotions/` on QTRP. After that you can play your video with  `/qt_robot/emotion/show` and you should pass your filename without .avi extension. For example we have animation `happy.avi` in folder `/home/qtrobot/robot/data/emotions/QT` so my command would be:
```bash
rosservice call /qt_robot/emotion/show "name: 'QT/happy'"
```
If you want to show some images you need to create .avi file to show them and use the same emotion interface.

aerith-zkelov, 2021-09-13:
> Hi @aerith-zkelov,
> The simple way is to use qtrobot emotion interface, which can only play .avi files. Copy your video to `/home/qtrobot/robot/data/emotions/` on QTRP. After that you can play your video with `/qt_robot/emotion/show` and you should pass your filename without .avi extension. For example we have animation `happy.avi` in folder `/home/qtrobot/robot/data/emotions/QT` so my command would be:
> 
> ```shell
> rosservice call /qt_robot/emotion/show "name: 'QT/happy'"
> ```
> 
> If you want to show some images you need to create .avi file to show them and use the same emotion interface.

noted, thanks for the advice!

aerith-zkelov, 2021-11-02:
> Hi @aerith-zkelov, The simple way is to use qtrobot emotion interface, which can only play .avi files. Copy your video to `/home/qtrobot/robot/data/emotions/` on QTRP. After that you can play your [...]

Hi based on my current knowledge we should send the file using SCP right? Since we can't directly access QTRP from the robot. Or is there another way that i am not aware of?

dkcodeguy, 2021-11-02:
HI @aerith-zkelov,
Yes, the easiest way would be using SCP. You can use SCP form QTPC or from your own computer. 
You can check our [documentation](https://docs.luxai.com/docs/modules/display#creating-and-playing-your-own-emotion) on how to display custom emotions.