luxai-qtrobot/QA #44: google cloud speech recognition does not work

ZouJennie, 2021-05-04:
Hello, it's me again.

After strictly following the step-by-step installation of the speech recognition tutorial on your github, I met a problem.

Just after installation, everything worked well. I could get transcript of rosservice /qt_robot/speech/recognize. But after some times of test, now i can just get `transcript: ' ' `

```
qtrobot@QT146:~$ rosservice call /qt_robot/speech/recognize "language: 'fr-FR'
options:
- ''
timeout: 10" 
transcript: ''
```

by the way, I also tried to install speech_recognition of python on QT respberry before, but it didn't work. There is a problem with jack. I don't know if there is some link between these two problem.

```
qtrobot@QT146:~$ python -m speech_recognition 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
A moment of silence, please...
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Set minimum energy threshold to 45.1318296212
Say something!
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock

```

Thank you for your attention

dkcodeguy, 2021-05-04:
Hi @ZouJennie,

Normally this warning for jack shouldn't be a problem, because we are not using that for the QTrobot microphone. Also, you should exactly specify which device you are using for speech recognition. For example, using pyaudio configuration for QTrobot microphone should be something like this: 
```
 # open mic audio device
    self.stream = self.paudio.open(
          start=False,
          format=pyaudio.paInt16,
          input_device_index=self.device['index'],
          channels=1,
          rate=16000,
          frames_per_buffer=int(16000 / 10),
          stream_callback=self.callback_stream,
          input=True)
```
What is important here is to find the correct device index. With pyaudio you can do this:
```
 """
 Get ReSpeaker microphone device
 """
    def get_respeaker_device(self):
        for i in range(self.paudio.get_host_api_count()):
            try:
                host = self.paudio.get_host_api_info_by_index(i)
                if host["name"]=="ALSA":
                    host_index = host['index']
                    for d in range(host['deviceCount']):
                        device = self.paudio.get_device_info_by_host_api_device_index(host_index, d)
                        if "ReSpeaker" in device['name']:
                            return device
            except:
                return None
        return None
```
In Linux, because of some privileges sometimes if you don't run your code with Sudo, it can't open the microphone port. So make sure if you are testing something you can try running it with Sudo.

I would like to know did you pull the latest code from Github and did you follow the [full-instalation](https://docs.luxai.com/examples/qt_gspeech_service/#2-full-installation-of-google-speech-interface) for google speech and are you running everything in virtualenv?

It is possible that because google speech needs an internet connection, it can happen that sometimes it loses the connection or doesn't receive the response back. You can try rebooting QTrobot, wifi or just or service.
Hope this clears some things up.

ZouJennie, 2021-05-07:
Yeah, after check there is a problem of internet connection. Robot connect unstably to internet. Many times, I saw robot connects to wifi but i couldn't open any website with its wifi QT146. Is there a solution for that? I can't reboot the robot during experiment.

closed 2021-05-31 by apaikan