luxai-qtrobot/QA #49: Unable to communicate with master

aerith-zkelov, 2021-09-13:
Hi, I'd like to ask a question on this issue I faced. I've been working with QTRobot using my laptop via ssh for quite some time now. However, today I tried to work directly with the QTPC and I tried to run a simple command and got this issue. May I know what may be causing this?

`qtrobot@QTPC:~$ rosservice call /qt_robot/speech/say "message: 'Hello.'"
ERROR: Unable to communicate with master!`

dkcodeguy, 2021-09-13:
HI @aerith-zkelov
Can you ping `QTRP` from QTPC? 
Sometimes the problem happens when QTrobot is not powered off correctly and QTRP losses connection to QTPC. You can try powering the QTrobot off and when is completely off  unplug it and plug it back on.

aerith-zkelov, 2021-09-13:
That worked thanks. And another question, I cloned tutorial files from the Github and encountered this issue, may i know what's causing it?
[image]

dkcodeguy, 2021-09-13:
Entire tutorial repo can't be build on QTPC, some of the stuff is just for QTRP.
As I can see from your log you have issue with `motors_custom_controller` that is just for QTRP. Copy just the example or demo that you need to catkin_ws on QTPC and try to build it

aerith-zkelov, 2021-09-13:
I see thank you for the clarification

closed 2021-09-13 by dkcodeguy