luxai-qtrobot/QA #51: QTRP user developer or qtrobot

sulibo, 2021-11-04:
In the newer version of QTrobot, QTRP is accessed from QTPC via ssh under the username developer instead of qtrobot. However, the auto start scripts and tutorials (for example [this one](https://github.com/luxai-qtrobot/software/blob/master/apps/qt_respeaker_app/autostart/start_qt_respeaker_app.sh)) are still based on the username qtrobot. Is it possible to update the relevant documents? Thanks!

dkcodeguy, 2021-11-04:
Hi @sulibo,
Everything that is running on the QTRP should be running under username 'qtrobot'. User 'developer' is just for ssh access, if you need to develop or change something you can use 'qtrobot' user. You can find more tutorials and tips on how to use QTrobot in our [documentation](https://docs.luxai.com/).

sulibo, 2021-11-04:
Hi @dkcodeguy, 
Thanks for the swift reply. Okay, I see. I assume the password for 'qtrobot' is the same as 'developer'.

dkcodeguy, 2021-11-04:
HI @sulibo,
Password for 'qtrobot' is always 'qtrobot'. If you change password of developer user that doesn't change qtrobot user.

sulibo, 2021-11-04:
Thanks. I failed to log in use 'qtrobot' as the password for the user 'qtrobot'. Probably, it was changed by my colleagues. I change it back and it works now.

closed 2021-11-04 by sulibo