luxai-qtrobot/QA #52: Motor node is crashing: Duration is out of dual 32-bit range

TechnoX, 2021-11-06:
Hi! 

So I've been working with the updated (Raspberry is running Buster) robot for a while now and the motor node is crashing with the following error:

`terminate called after throwing an instance of 'std::runtime_error'
  what():  Duration is out of dual 32-bit range`

I'm running qt_motor_advanced.launch from start_qt_motor.log on autostart. I attach the complete log-file to this post as well. 
[start_qt_motor.log](https://github.com/luxai-qtrobot/QA/files/7491042/start_qt_motor.log)

I got a similar error in June on the old robot running the qt_motor_advanced.launch : 

`terminate called after throwing an instance of 'std::runtime_error'
  what():  Time is out of dual 32-bit range`
See the complete error here: https://github.com/luxai-qtrobot/QA/issues/47#issuecomment-858174994

TechnoX, 2021-11-21:
I have done some more research on this. 

In one terminal I run "`roslaunch qt_motor qt_motor_advanced.launch`" and in another terminal I publish some messages to `/qt_robot/gesture/play`.

When I publish the first message, for example this: 
`rostopic pub --once /qt_robot/gesture/play std_msgs/String "data: 'QT/hi'"`
The robot execute the gesture without any problem. 

When I send the next gesture (after waiting for the first to complete first) with the exact same command: 
`rostopic pub --once /qt_robot/gesture/play std_msgs/String "data: 'QT/hi'"`
The robot executes the movement as before, but when it is done with the movement the motor node crashes with 

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Duration is out of dual 32-bit range
```

I get the same behavior when I test with my own gestures and with other built-in gestures, for example QT/send_kiss gives the same as above.

dkcodeguy, 2021-12-01:
Reason of the issue incompatibility of QTrobot libraries (64-bit system to 32-bit system).
We have fixed the issue and to update your QTrobot please contact us on support@luxai.com.

closed 2021-12-01 by dkcodeguy